MIT-SPARK/KISS-Matcher #1: Aboult Code Release

nuo-1, 2024-10-22:
Hi, your work KISS-Matcher is excellent. are there any plans to open source recently?

nnuli-111, 2024-10-23:
this is forked by mit
---- Replied Message ----
| From | Xiaoliang ***@***.***> |
| Date | 10/22/2024 23:18 |
| To | ***@***.***> |
| Cc | ***@***.***> |
| Subject | Re: [MIT-SPARK/KISS-Matcher] Aboult Code Release (Issue #1) |

+1

LimHyungTae, 2024-10-25:
Dear all, appreciate your kind interest! Once it is accepted in ICRA'25, I'll upload the codes right away!

fdila, 2025-02-12:
Hi, I've seen a post on linkedin confirming that it was accepted at ICRA, congratulations!
Any updates on the release of the code?

NothingCD, 2025-02-12:
您好，邮件已收到，尽快给您回复。

LimHyungTae, 2025-02-16:
@fdila I'm almost done! Hold on.

LimHyungTae, 2025-02-17:
@fdila @victhormoreno @Andymulb @samcts-auki @NothingCD @narutojxl @nuo-1 

There you go! Codes have been uploaded.