MIT-SPARK/KISS-Matcher #10: License & ROS2 Example

ArmanAstudSeevia, 2025-02-18:
Hello everyone,

Thank you for sharing such an amazing project.
We have 2 main questions:

1. Under what license are you sharing the project, MIT, Apache-2?
2. I have not been able to see a ros2-node example, the examples are only for PCD files. Do you plan to update these examples?

Thanks.

LimHyungTae, 2025-02-18:
A1. We are pursuing an MIT license, but there might be a license issue. Please let us know!

A2. I'm working on setting up a SLAM back-end with KISS-Matcher (using ROS1 code here: https://github.com/engcang/FAST-LIO-SAM-QN) and ROS2 example.

ArmanAstudSeevia, 2025-02-18:
Thank you for addressing the issue.

1. We can now see the license in the project root! :)
2. Great! We’ll stay tuned to test it as soon as possible.

Closing the issue.

closed 2025-02-18 by ArmanAstudSeevia