MIT-SPARK/KISS-Matcher #9: [BUG] Teaser-PlusPlus - broken link in submodule

Josgonmar, 2025-02-18:
Hi there,

First of all, thank you for open-sourcing such an amazing project! I was excited to test it with some of my own point clouds, but when I tried to compile the examples, I noticed that the submodule appears to be broken. I know it’s an easy fix, but I just wanted to bring it to your attention. 😄

Best regards!

LimHyungTae, 2025-02-18:
At the pipeline, there's no depdendency in TEASER++. Did you run `examples`?

Josgonmar, 2025-02-18:
Yes. I tried to compile the examples, and teaser++ is defined as a dependency in the CMakeLists.txt there. I don't know if it's actually used in the code, but it's there 100%

My question was more related to the empty TEASER-plusplus folder inside _cpp/examples/install_.
Although checking it out again, the recommended installation script inside _shellscripts_ clones the package where this script is located.

LimHyungTae, 2025-02-18:
Oh, man. I wrongly push that one. Did you run this command? 

[image]

LimHyungTae, 2025-02-18:
Hmm....I found that `./run_matching_using_bunny` showed below error message when I tried it in a docker:

```
./run_matching_using_bunny: error while loading shared libraries: libpmc.so: cannot open shared object file: No such file or directory
```

Let me fix soon.

Josgonmar, 2025-02-18:
Yeah, I was able to successfully compile and run the examples, at least ./kiss_matcher_run.

I'm closing this issue now since, as you mentioned, the TEASER-plusplus folder was mistakenly pushed to the repository 😅. That was the only thing I wanted to verify.

Thanks!

PD:
> Hmm....I found that `./run_matching_using_bunny` showed below error message when I tried it in a docker:
> 
> ```
> ./run_matching_using_bunny: error while loading shared libraries: libpmc.so: cannot open shared object file: No such file or directory
> ```
> 
> Let me fix it soon.

If the lib is already installed, perhaps `sudo ldconfig` fixes it?

closed 2025-02-18 by Josgonmar

LimHyungTae, 2025-02-18:
My pleasure! Thanks for letting me know. I was rush, so some parts were not cleaned out! Sorry for the inconvenience ;)

LimHyungTae, 2025-02-18:
and `ldconfig` resolves that issue, bro!