MIT-SPARK/KISS-Matcher #24: [QUESTION] Confidence information using scan to scan matching

hofbauerc, 2025-02-21:
Hi,

Thank you for KISS-Matcher. I already tried it with some self recorded data and it seems to work pretty well. 

However, one thing I am not quite sure about: Is it possible to get any feedback on the confidence on the matching procedure. In the cpp example code for scan matching, the RegistrationSolution class only contains a valid or not valid bool. Is there any way to get information if the matching has been achieved with some kind of confidence?

Maybe I am missing something in the code regarding this feature. 

I look forward to your answer.

LimHyungTae, 2025-02-21:
Hi @hofbauerc , if you check the # of inliers, it can be used as a verification (e.g., mostly, failed registration outputs few final inliers when being checked by `solver_->getTranslationInliers().size()`)

Oh man, but I've realized there's no function to get the final number of inliers. If it is possible, can you do PR for me?

hofbauerc, 2025-02-21:
@LimHyungTae ok so basically the amount of final inliers (rot and trans) specifiy how good the matching could be achieved. I need to test that to see if that is a good way to tell the confidence of the match. When I am done testing I will add the method to the KISSMatcher class. I try to add this next week.

hofbauerc, 2025-02-21:
Did some further testing and checked the paper again regarding the inlies. Added a pull request for the new function #26

However, the solver is still marked as a public variable which could be changed to private but I did not want to change your code that much.

LimHyungTae, 2025-02-21:
Oh got it. It should be private. Thanks!

LimHyungTae, 2025-02-23:
so, now you can verify whether the estimation succeeded or failed like this example:

[image]

LimHyungTae, 2025-02-23:
If you have any additional concerns, please reopen this issue!