MIT-SPARK/KISS-Matcher #25: GPU/CUDA support for Python

jose-donate, 2025-02-21:
Hi.

Thanks for sharing the results of your research. So far, I have tried the KISS matcher with my own data and it has worked beautifully yet, when launching the run_kiss_matcher.py with some heavy point clouds, I can see the hardware monitor go up to 100% CPU usage and 0% GPU usage. 

I tried installing FLANN matcher with CUDA enabled but that didn't solve the problem. Is there something I may be missing that will actually leverage the GPU available on my computer?

I'm working on WSL2 and have a NVidia RTX 4070 Ti.

Thanks in advance and keep up the good work.

jose-donate, 2025-02-21:
They are slightly over 4M points, which isn't that big actually, yet it takes a while performing the matching. This is the output, if you want more details.

```
Loaded source point cloud: (4376621, 3)
Loaded target point cloud: (4386873, 3)
============== Time ==============
Voxelization: 0.109465 sec
Extraction  : 12.3635 sec
Pruning     : 0.019784 sec
Matching    : 94.308 sec
Solving     : 0.000730053 sec
----------------------------------
Total     : 106.802 sec
```

I'll give it a try to the voxel size. 

Appreciate your fast response.

LimHyungTae, 2025-02-21:
Wow, that's huge, and I guess it's almost a city-level point cloud. In that case, just put a larger voxel size ( > 5.0m) and make sure that the # of points becomes hundreds of thousands of points.

LimHyungTae, 2025-02-21:
By the way, does registration succeed? Could you also share the visualized results please?

closed 2025-02-25 by LimHyungTae