MIT-SPARK/KISS-Matcher #37: [BUG] ImportError: libkiss_matcher_core.so: cannot open shared object file: No such file or directory

ttsesm, 2025-03-20:
Hi @LimHyungTae,

Thanks for the library, I am trying to install the python bindings of the KISS-Matcher but while I am following the instructions from [here](https://github.com/MIT-SPARK/KISS-Matcher/tree/main/python) and everything seems to go well when I am trying to run a script where I am importing the `kiss_matcher` module I am getting the following error:
```
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 666, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 565, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1173, in create_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
ImportError: libkiss_matcher_core.so: cannot open shared object file: No such file or directory
```
Any, idea what could be the issue and how I could possibly resolve it... 🤔 

Thanks.

LimHyungTae, 2025-03-20:
Could you run `sudo ldconfig`?

ttsesm, 2025-03-20:
Yup, it should be working....

The output of the `pip3 install -e . --verbose` is the following:
```
$ pip3 install -e . --verbose
Using pip 25.0.1 from /home/ttsesm/Development/cameraPose/venv_camera_pose/lib/python3.9/site-packages/pip (python 3.9)
Obtaining file:///home/ttsesm/Development/cameraPose/KISS-Matcher/python
  Running command pip subprocess to install build dependencies
  Using pip 25.0.1 from /home/ttsesm/Development/cameraPose/venv_camera_pose/lib/python3.9/site-packages/pip (python 3.9)
  Collecting scikit_build_core
    Obtaining dependency information for scikit_build_core from https://files.pythonhosted.org/packages/0a/ba/b37b9802f503894a46ef34aaa5851344cde48b39ab0af5057a6ee4f0d631/scikit_build_core-0.11.0-py3-none-any.whl.metadata
    Using cached scikit_build_core-0.11.0-py3-none-any.whl.metadata (21 kB)
  Collecting pybind11
    Obtaining dependency information for pybind11 from https://files.pythonhosted.org/packages/13/2f/0f24b288e2ce56f51c920137620b4434a38fd80583dbbe24fc2a1656c388/pybind11-2.13.6-py3-none-any.whl.metadata
    Using cached pybind11-2.13.6-py3-none-any.whl.metadata (9.5 kB)
  Collecting exceptiongroup>=1.0 (from scikit_build_core)
    Obtaining dependency information for exceptiongroup>=1.0 from https://files.pythonhosted.org/packages/02/cc/b7e31358aac6ed1ef2bb790a9746ac2c69bcb3c8588b41616914eb106eaf/exceptiongroup-1.2.2-py3-none-any.whl.metadata
    Using cached exceptiongroup-1.2.2-py3-none-any.whl.metadata (6.6 kB)
  Collecting packaging>=21.3 (from scikit_build_core)
    Obtaining dependency information for packaging>=21.3 from https://files.pythonhosted.org/packages/88/ef/eb23f262cca3c0c4eb7ab1933c3b1f03d021f2c48f54763065b6f0e321be/packaging-24.2-py3-none-any.whl.metadata
    Using cached packaging-24.2-py3-none-any.whl.metadata (3.2 kB)
  Collecting pathspec>=0.10.1 (from scikit_build_core)
    Obtaining dependency information for pathspec>=0.10.1 from https://files.pythonhosted.org/packages/cc/20/ff623b09d963f88bfde16306a54e12ee5ea43e9b597108672ff3a408aad6/pathspec-0.12.1-py3-none-any.whl.metadata
    Using cached pathspec-0.12.1-py3-none-any.whl.metadata (21 kB)
  Collecting tomli>=1.2.2 (from scikit_build_core)
    Obtaining dependency information for tomli>=1.2.2 from https://files.pythonhosted.org/packages/6e/c2/61d3e0f47e2b74ef40a68b9e6ad5984f6241a942f7cd3bbfbdbd03861ea9/tomli-2.2.1-py3-none-any.whl.metadata
    Using cached tomli-2.2.1-py3-none-any.whl.metadata (10 kB)
  Using cached scikit_build_core-0.11.0-py3-none-any.whl (179 kB)
  Using cached pybind11-2.13.6-py3-none-any.whl (243 kB)
  Using cached exceptiongroup-1.2.2-py3-none-any.whl (16 kB)
  Using cached packaging-24.2-py3-none-any.whl (65 kB)
  Using cached pathspec-0.12.1-py3-none-any.whl (31 kB)
  Using cached tomli-2.2.1-py3-none-any.whl (14 kB)
  Installing collected packages: tomli, pybind11, pathspec, packaging, exceptiongroup, scikit_build_core
    Creating /tmp/pip-build-env-p5zx967h/overlay/bin
    changing mode of /tmp/pip-build-env-p5zx967h/overlay/bin/pybind11-config to 755
  Successfully installed exceptiongroup-1.2.2 packaging-24.2 pathspec-0.12.1 pybind11-2.13.6 scikit_build_core-0.11.0 tomli-2.2.1
  Installing build dependencies ... done
  Running command Checking if build backend supports build_editable
  Checking if build backend supports build_editable ... done
  Running command Getting requirements to build editable
  Could not determine CMake version via --version, got '' 'Traceback (most recent call last):\n  File "/home/ttsesm/Development/cameraPose/venv_camera_pose/bin/cmake", line 5, in <module>\n    from cmake import cmake\nModuleNotFoundError: No module named \'cmake\'\n'
  Getting requirements to build editable ... done
  Running command pip subprocess to install backend dependencies
  Using pip 25.0.1 from /home/ttsesm/Development/cameraPose/venv_camera_pose/lib/python3.9/site-packages/pip (python 3.9)
  Collecting cmake>=3.16
    Obtaining dependency information for cmake>=3.16 from https://files.pythonhosted.org/packages/59/e8/096984b89133681533650b9078c5ed1c5c9b534e869b5487f22d4de1935c/cmake-3.31.6-py3-none-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata
    Using cached cmake-3.31.6-py3-none-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (6.3 kB)
  Using cached cmake-3.31.6-py3-none-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (27.8 MB)
  Installing collected packages: cmake
    Creating /tmp/pip-build-env-p5zx967h/normal/bin
    changing mode of /tmp/pip-build-env-p5zx967h/normal/bin/ccmake to 755
    changing mode of /tmp/pip-build-env-p5zx967h/normal/bin/cmake to 755
    changing mode of /tmp/pip-build-env-p5zx967h/normal/bin/cpack to 755
    changing mode of /tmp/pip-build-env-p5zx967h/normal/bin/ctest to 755
  Successfully installed cmake-3.31.6
  Installing backend dependencies ... done
  Running command Preparing editable metadata (pyproject.toml)
  *** scikit-build-core 0.11.0 using CMake 3.31.6 (metadata_editable)
  Preparing editable metadata (pyproject.toml) ... done
Requirement already satisfied: numpy in /home/ttsesm/Development/cameraPose/venv_camera_pose/lib/python3.9/site-packages (from kiss_matcher==1.0.0) (1.26.4)
Requirement already satisfied: open3d in /home/ttsesm/Development/cameraPose/venv_camera_pose/lib/python3.9/site-packages (from kiss_matcher==1.0.0) (0.19.0)
[... cut ...]
  Created wheel for kiss_matcher: filename=kiss_matcher-1.0.0-cp39-cp39-linux_x86_64.whl size=1487809 sha256=0e6d59b8d19679af81d1b1f51a56a0cd9053345ea0094d40d607e2bfcfb79656
  Stored in directory: /tmp/pip-ephem-wheel-cache-er61kmys/wheels/73/45/13/016ab7d22110e145f6fad6f81afde5a51d1a7b58dbeaeaf67f
Successfully built kiss_matcher
Installing collected packages: kiss_matcher
Successfully installed kiss_matcher-1.0.0
```

which everthing indicates that there were not issues... 🤔 

I've tried the `sudo ldconfig` solution, no success though.
Any other idea? Where is the `libkiss_matcher_core.so` created? Can I link it externally, somehow?

ttsesm, 2025-03-21:
Problem solved with a fresh installation.

Thanks.

ttsesm, 2025-03-21:
@LimHyungTae sure. The issue, came from the fact that I am using a linux distribution (Arch linux) which has a different structure policy than Ubuntu (binaries and libs are not installed in `/usr/local`). So, my guess is that when you run `pip3 install -e . --verbose` though it seems not to show any issues, it couldn't find the lib even after running `ldconfig`.  

Applying a new fresh installation in an Ubuntu machine, everything worked fine. My guess is that, what you could possibly do is to create a local inclusion of the compiled lib as I think you do with the TEASER++ compilation/installation.

LimHyungTae, 2025-03-21:
I’m always torn between installing the library locally or installing it via add_subdirectory(${CMAKE_CURRENT_SOURCE_DIR}/../cpp/kiss_matcher ${CMAKE_CURRENT_BINARY_DIR}/kiss_matcher). 

I’ll give it some thought (or update package)!

bnkf1156f, 2025-04-25:
Using WSL2 Ubuntu 22.04.5 LTS

I solved my issue by firstly finding where _libkiss_matcher_core.so_ and in my case _/home/virkha/.local/lib/python3.10/site-packages/lib_ 

Then exporting the environment variable:
`export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/virkha/.local/lib/python3.10/site-packages/lib`

And now do run any example as I did=>
```
(kiss_matcher) virkha@PF2DMKGC:~/KISS-Matcher/python$ python3 examples/run_kiss_matcher.py --src_path data/Vel16/src.pcd --tgt_path data/Vel16/tgt.pcd --resolution 0.2
Loaded source point cloud: (25370, 3)
Loaded target point cloud: (25049, 3)
Resolution: 0.2
Yaw Augmentation Angle: None
============== Time ==============
Voxelization: 0.00683609 sec
Extraction  : 0.0351304 sec
Pruning     : 0.0122404 sec
Matching    : 0.03717 sec
Solving     : 0.000100396 sec
----------------------------------
Total     : 0.0914773 sec
====== # of correspondences ======
# initial pairs : 824
# pruned pairs  : 40
----------------------------------
# rot inliers   : 26
# trans inliers : 21
==================================
=> Registration likely succeeded XD
```

I hope it helps 💯

LimHyungTae, 2025-04-29:
Thanks a lot! This error really really looks off, because in Ubuntu 20.04, this error doesn't happen. I'll check it ASAP and fix it.

LimHyungTae, 2025-04-29:
Hmm...I went back to #19, and it worked well without any ImportError. I guess I made some mistakes while maintaining the repository :( 

The main difference is, in #19, the following files are installed:

```
./lib/python3.10/site-packages/__pycache__/_kiss_matcher_editable.cpython-310.pyc
./lib/python3.10/site-packages/kiss_matcher.cpython-310-x86_64-linux-gnu.so
./lib/python3.10/site-packages/_kiss_matcher_editable.py
./lib/python3.10/site-packages/_kiss_matcher_editable.pth
./lib/python3.10/site-packages/lib/cmake/kiss_matcher/kiss_matcherConfig.cmake
./lib/python3.10/site-packages/lib/cmake/kiss_matcher/kiss_matcherTargets.cmake
./lib/python3.10/site-packages/lib/cmake/kiss_matcher/kiss_matcherTargets-release.cmake
**./lib/python3.10/site-packages/lib/libkiss_matcher_core.a**
./lib/python3.10/site-packages/include/kiss_matcher/ROBINMatching.cpp
./lib/python3.10/site-packages/include/kiss_matcher/KISSMatcher.cpp
./lib/python3.10/site-packages/include/kiss_matcher/points/eigen.hpp
./lib/python3.10/site-packages/include/kiss_matcher/points/downsampling.hpp
./lib/python3.10/site-packages/include/kiss_matcher/points/traits.hpp
./lib/python3.10/site-packages/include/kiss_matcher/points/fast_floor.hpp
./lib/python3.10/site-packages/include/kiss_matcher/points/point_cloud.hpp
./lib/python3.10/site-packages/include/kiss_matcher/points/vector3i_hash.hpp
./lib/python3.10/site-packages/include/kiss_matcher/FasterPFH.cpp
./lib/python3.10/site-packages/include/kiss_matcher/KISSMatcher.hpp
./lib/python3.10/site-packages/include/kiss_matcher/tsl/robin_hash.h
./lib/python3.10/site-packages/include/kiss_matcher/tsl/robin_map.h
./lib/python3.10/site-packages/include/kiss_matcher/tsl/robin_growth_policy.h
./lib/python3.10/site-packages/include/kiss_matcher/tsl/robin_set.h
./lib/python3.10/site-packages/include/kiss_matcher/GncSolver.hpp
./lib/python3.10/site-packages/include/kiss_matcher/FasterPFH.hpp
./lib/python3.10/site-packages/include/kiss_matcher/ROBINMatching.hpp
./lib/python3.10/site-packages/include/kiss_matcher/GncSolver.cpp
./lib/python3.10/site-packages/include/kiss_matcher/kdtree/nanoflann.hpp
./lib/python3.10/site-packages/include/kiss_matcher/kdtree/traits.hpp
./lib/python3.10/site-packages/include/kiss_matcher/kdtree/kdtree.hpp
./lib/python3.10/site-packages/include/kiss_matcher/kdtree/kdtree_tbb.hpp
./lib/python3.10/site-packages/include/kiss_matcher/kdtree/nanoflann_tbb.hpp
./lib/python3.10/site-packages/kiss_matcher-1.0.0.dist-info/INSTALLER
./lib/python3.10/site-packages/kiss_matcher-1.0.0.dist-info/RECORD
./lib/python3.10/site-packages/kiss_matcher-1.0.0.dist-info/REQUESTED
./lib/python3.10/site-packages/kiss_matcher-1.0.0.dist-info/METADATA
./lib/python3.10/site-packages/kiss_matcher-1.0.0.dist-info/direct_url.json
./lib/python3.10/site-packages/kiss_matcher-1.0.0.dist-info/WHEEL
./lib/python3.10/site-packages/kiss_matcher-1.0.0.dist-info/licenses/LICENSE
```

Whereas, the current version installed below files:

```
./lib/python3.10/site-packages/__pycache__/_kiss_matcher_editable.cpython-310.pyc
./lib/python3.10/site-packages/kiss_matcher.cpython-310-x86_64-linux-gnu.so
./lib/python3.10/site-packages/_kiss_matcher_editable.py
./lib/python3.10/site-packages/_kiss_matcher_editable.pth
./lib/python3.10/site-packages/lib/cmake/kiss_matcher/kiss_matcherConfig.cmake
./lib/python3.10/site-packages/lib/cmake/kiss_matcher/kiss_matcherTargets.cmake
./lib/python3.10/site-packages/lib/cmake/kiss_matcher/kiss_matcherTargets-release.cmake
**./lib/python3.10/site-packages/lib/libkiss_matcher_core.so**
./lib/python3.10/site-packages/share/doc/kiss_matcher/README.md
./lib/python3.10/site-packages/include/kiss_matcher/ROBINMatching.cpp
./lib/python3.10/site-packages/include/kiss_matcher/KISSMatcher.cpp
./lib/python3.10/site-packages/include/kiss_matcher/points/eigen.hpp
./lib/python3.10/site-packages/include/kiss_matcher/points/downsampling.hpp
./lib/python3.10/site-packages/include/kiss_matcher/points/traits.hpp
./lib/python3.10/site-packages/include/kiss_matcher/points/fast_floor.hpp
./lib/python3.10/site-packages/include/kiss_matcher/points/point_cloud.hpp
./lib/python3.10/site-packages/include/kiss_matcher/points/vector3i_hash.hpp
./lib/python3.10/site-packages/include/kiss_matcher/FasterPFH.cpp
./lib/python3.10/site-packages/include/kiss_matcher/KISSMatcher.hpp
./lib/python3.10/site-packages/include/kiss_matcher/tsl/robin_hash.h
./lib/python3.10/site-packages/include/kiss_matcher/tsl/robin_map.h
./lib/python3.10/site-packages/include/kiss_matcher/tsl/robin_growth_policy.h
./lib/python3.10/site-packages/include/kiss_matcher/tsl/robin_set.h
./lib/python3.10/site-packages/include/kiss_matcher/GncSolver.hpp
./lib/python3.10/site-packages/include/kiss_matcher/FasterPFH.hpp
./lib/python3.10/site-packages/include/kiss_matcher/ROBINMatching.hpp
./lib/python3.10/site-packages/include/kiss_matcher/GncSolver.cpp
./lib/python3.10/site-packages/include/kiss_matcher/kdtree/nanoflann.hpp
./lib/python3.10/site-packages/include/kiss_matcher/kdtree/traits.hpp
./lib/python3.10/site-packages/include/kiss_matcher/kdtree/kdtree.hpp
./lib/python3.10/site-packages/include/kiss_matcher/kdtree/kdtree_tbb.hpp
./lib/python3.10/site-packages/include/kiss_matcher/kdtree/nanoflann_tbb.hpp
./lib/python3.10/site-packages/kiss_matcher-1.0.0.dist-info/INSTALLER
./lib/python3.10/site-packages/kiss_matcher-1.0.0.dist-info/RECORD
./lib/python3.10/site-packages/kiss_matcher-1.0.0.dist-info/REQUESTED
./lib/python3.10/site-packages/kiss_matcher-1.0.0.dist-info/METADATA
./lib/python3.10/site-packages/kiss_matcher-1.0.0.dist-info/direct_url.json
./lib/python3.10/site-packages/kiss_matcher-1.0.0.dist-info/WHEEL
./lib/python3.10/site-packages/kiss_matcher-1.0.0.dist-info/licenses/LICENSE
```

Main difference would be the `libkiss_matcher_core.a` and `libkiss_matcher_core.so`.

Let me check them further!

bnkf1156f, 2025-04-29:
Thank you for looking into this issue!😄

LimHyungTae, 2025-04-29:
It was my fault. Now it has been fixed. Thanks! (see #59)

closed 2025-04-29 by LimHyungTae

bnkf1156f, 2025-04-30:
Thank you for solving this issue!