MIT-SPARK/KISS-Matcher #38: [BUG] Compatibility issue with TBB 2022: tbb::task moved to detail::d1 namespace

luoxinyong, 2025-03-22:
Hi, when building with TBB 2022.0, the compiler reports errors about tbb::task not being declared. The legacy TBB interfaces seem to have been moved to tbb::detail::d1 in newer TBB versions. Could you update the namespace references to support modern TBB?
here is my error：
/usr/include/c++/10/pstl/parallel_backend_tbb.h:70:10: error: ‘tbb::task’ has not been declared
   70 |     tbb::task::self().group()->cancel_group_execution();
      |          ^~~~
/usr/include/c++/10/pstl/parallel_backend_tbb.h: At global scope:
/usr/include/c++/10/pstl/parallel_backend_tbb.h:414:1: error: expected class-name before ‘{’ token
  414 | {
      | ^
/usr/include/c++/10/pstl/parallel_backend_tbb.h:415:23: error: ‘task’ in namespace ‘tbb’ does not name a type
  415 |     /*override*/ tbb::task*
      |                       ^~~~
/usr/include/c++/10/pstl/parallel_backend_tbb.h:438:6: error: ‘task’ in namespace ‘tbb’ does not name a type
  438 | tbb::task*
      |      ^~~~
/usr/include/c++/10/pstl/parallel_backend_tbb.h:483:1: error: expected class-name before ‘{’ token
  483 | {
      | ^
/usr/include/c++/10/pstl/parallel_backend_tbb.h:490:23: error: ‘task’ in namespace ‘tbb’ does not name a type
  490 |     /*override*/ tbb::task*
      |                       ^~~~
/usr/include/c++/10/pstl/parallel_backend_tbb.h:520:6: error: ‘task’ in namespace ‘tbb’ does not name a type
  520 | tbb::task*
      |      ^~~~
/usr/include/c++/10/pstl/parallel_backend_tbb.h: In lambda function:
/usr/include/c++/10/pstl/parallel_backend_tbb.h:602:24: error: ‘task’ has not been declared in ‘tbb’
  602 |             using tbb::task;
      |                        ^~~~
/usr/include/c++/10/pstl/parallel_backend_tbb.h:603:13: error: ‘task’ has not been declared
  603 |             task::spawn_root_and_wait(*new (task::allocate_root())
      |             ^~~~
/usr/include/c++/10/pstl/parallel_backend_tbb.h:603:45: error: ‘task’ was not declared in this scope; did you mean ‘tbb::detail::d1::task’?
  603 |             task::spawn_root_and_wait(*new (task::allocate_root())
      |                                             ^~~~
      |                                             tbb::detail::d1::task

LimHyungTae, 2025-03-24:
Could you please test it with

```
option(USE_SYSTEM_TBB "Use system pre-installed oneAPI/tbb" ON)
```

?

luoxinyong, 2025-03-25:
I have tried the modification you mentioned on my WSL, i.e., option(USE_SYSTEM_TBB "Use system pre-installed oneAPI/tbb" ON), but it did not resolve my issue. However, after reinstalling a fresh version of Ubuntu 22.04, Kiss_Matcher was successfully installed on my WSL (Ubuntu 22.04) and tested on my dataset.
I am not sure if the failure to install Kiss_Matcher was due to the excessive environment configurations in my system (since I research SLAM and have made many modifications to my system's configuration). Once again, thank you for your response. I will continue trying to install Kiss_Matcher on my WSL (Ubuntu 20.04).

Here is my error after（USE_SYSTEM_TBB "Use system pre-installed oneAPI/tbb" ON)：
In file included from /usr/include/c++/10/pstl/parallel_backend.h:16,
                 from /usr/include/c++/10/pstl/algorithm_impl.h:22,
                 from /usr/include/c++/10/pstl/glue_execution_defs.h:50,
                 from /usr/include/c++/10/execution:32,
                 from /home/bnxy/kiss_matcher/KISS-Matcher/cpp/kiss_matcher/core/kiss_matcher/ROBINMatching.hpp:11,
                 from /home/bnxy/kiss_matcher/KISS-Matcher/cpp/kiss_matcher/core/kiss_matcher/ROBINMatching.cpp:9:
/usr/include/c++/10/pstl/parallel_backend_tbb.h: In function ‘void __pstl::__par_backend::__cancel_execution()’:
/usr/include/c++/10/pstl/parallel_backend_tbb.h:70:10: error: ‘tbb::task’ has not been declared
   70 |     tbb::task::self().group()->cancel_group_execution();
      |          ^~~~
/usr/include/c++/10/pstl/parallel_backend_tbb.h: At global scope:
/usr/include/c++/10/pstl/parallel_backend_tbb.h:414:1: error: expected class-name before ‘{’ token
  414 | {
      | ^
/usr/include/c++/10/pstl/parallel_backend_tbb.h:415:23: error: ‘task’ in namespace ‘tbb’ does not name a type
  415 |     /*override*/ tbb::task*
      |                       ^~~~
/usr/include/c++/10/pstl/parallel_backend_tbb.h:438:6: error: ‘task’ in namespace ‘tbb’ does not name a type
  438 | tbb::task*
      |      ^~~~
/usr/include/c++/10/pstl/parallel_backend_tbb.h:483:1: error: expected class-name before ‘{’ token
  483 | {
      | ^
/usr/include/c++/10/pstl/parallel_backend_tbb.h:490:23: error: ‘task’ in namespace ‘tbb’ does not name a type
  490 |     /*override*/ tbb::task*
      |                       ^~~~
/usr/include/c++/10/pstl/parallel_backend_tbb.h:520:6: error: ‘task’ in namespace ‘tbb’ does not name a type
  520 | tbb::task*
      |      ^~~~
/usr/include/c++/10/pstl/parallel_backend_tbb.h: In lambda function:
/usr/include/c++/10/pstl/parallel_backend_tbb.h:602:24: error: ‘task’ has not been declared in ‘tbb’
  602 |             using tbb::task;
      |                        ^~~~
/usr/include/c++/10/pstl/parallel_backend_tbb.h:603:13: error: ‘task’ has not been declared
  603 |             task::spawn_root_and_wait(*new (task::allocate_root())
      |             ^~~~
/usr/include/c++/10/pstl/parallel_backend_tbb.h:603:45: error: ‘task’ was not declared in this scope; did you mean ‘tbb::detail::d1::task’?
  603 |             task::spawn_root_and_wait(*new (task::allocate_root())
      |                                             ^~~~
      |                                             tbb::detail::d1::task

ttsesm, 2025-03-31:
> Could you please test it with
> 
> ```
> option(USE_SYSTEM_TBB "Use system pre-installed oneAPI/tbb" ON)
> ```
> 
> in `cpp/kiss_matcher/CMakeLists.txt`?

This worked for me.

closed 2025-03-31 by LimHyungTae