MIT-SPARK/KISS-Matcher #39: [QUESTION] Improving KISS-Matcher registration result

ttsesm, 2025-03-24:
Hi @LimHyungTae,

I would like to ask you if possibly I could improve the result of the KISS-Matcher. I have the following 3 point clouds:
1st:
[image]
2nd:
[image]
3rd:
[image]

Which I would like to register. I am downsampling my pcds on my own using open3d's `voxel_down_sample()` functionality and disabling the `use_voxel_sampling` parameter in the `KISSMatcherConfig`. Which after playing a bit with this parameter I've noticed that it plays quite a significant role.

The result is the following which is really close but not perfect:
1st-2nd
[image]
1st-3rd
[image]
2nd-3rd
[image]

Thus, the question is how I could possibly improve the final result. I consider that the pcds are not that complicated with quite some common big surface overlapping. So I would consider them a relatively easy use case. Also I am not sure if I could possibly use some color information as assistance... 🤔 (it is quite interesting that there is almost nothing in the literature to take advantage of the color information)

Also I am not sure what I could possibly modify of the following parameters in order to help:
```
"voxel_size"_a                  = 0.3,
"use_voxel_sampling"_a          = true,
"use_quatro"_a                  = false,
"thr_linearity"_a               = 1.0,
"num_max_corr"_a                = 5000,
"normal_r_gain"_a               = 3.0,
"fpfh_r_gain"_a                 = 5.0,
"robin_noise_bound_gain"_a      = 1.0,
"solver_noise_bound_gain"_a     = 0.75,
"enable_noise_bound_clamping"_a = true)
```

Any feedback would be appreciated.
Thanks.

LimHyungTae, 2025-03-24:
First, could you lower the voxel size to 0.05 or 0.075 and rerun your code?

ttsesm, 2025-03-24:
Hhhmmm, well it is a bit unstable... 🤔 

For example in the following 5 scenes (https://filesender.garr.it/?s=download&token=851b157e-9cd8-4ab6-9454-18751e01dd33), I could make it work only once (scene 4):
Scene1:
[image]
Scene2:
[image]
Scene3:
[image]
Scene4:
[image]
Scene5:
[image]

and here is the code I am using (I am also applying some icp refinement at the end):
```
import kiss_matcher
import numpy as np
import open3d as o3d  # Only for the load of pcd and visualization

if __name__ == "__main__":

    print("1. Load pcds.")
    src_o3d = o3d.io.read_point_cloud("data/scene1/zed_1.ply")
    tgt_o3d = o3d.io.read_point_cloud("data/scene1/zed_2.ply")
    rlsense_o3d = o3d.io.read_point_cloud("data/scene1/L515.ply")

    print("2. Downsample pcds.")
    source_down = src_o3d.voxel_down_sample(0.008)
    target_down = tgt_o3d.voxel_down_sample(0.008)
    rlsense_down = rlsense_o3d.voxel_down_sample(0.008)

    print("3. Estimate normals.")
    source_down.estimate_normals(
        o3d.geometry.KDTreeSearchParamHybrid(radius=0.005 * 2, max_nn=30))
    target_down.estimate_normals(
        o3d.geometry.KDTreeSearchParamHybrid(radius=0.005 * 2, max_nn=30))
    rlsense_down.estimate_normals(
        o3d.geometry.KDTreeSearchParamHybrid(radius=0.005 * 2, max_nn=30))

    print("4. Setup KISS-Matcher.")
    params = kiss_matcher.KISSMatcherConfig(0.05, use_voxel_sampling=False)
    matcher = kiss_matcher.KISSMatcher(params)

    print("5. Run KISS-Matcher between 1 and 2.")
    result = matcher.estimate(source_down.points, target_down.points)
    matcher.print()

    num_rot_inliers = matcher.get_num_rotation_inliers()
    num_final_inliers = matcher.get_num_final_inliers()
    # NOTE(hlim): By checking the final inliers, we can determine whether
    # the registration was successful or not. The larger the threshold,
    # the more conservatively the decision is made.
    # See https://github.com/MIT-SPARK/KISS-Matcher/issues/24
    thres_num_inliers = 5
    if (num_final_inliers < thres_num_inliers):
        print("\033[1;33m=> Registration might have failed :(\033[0m\n")
    else:
        print("\033[1;32m=> Registration likely succeeded XD\033[0m\n")

    print("6. Run KISS-Matcher between 2 and 3.")
    result_rl = matcher.estimate(rlsense_down.points, target_down.points)
    matcher.print()

    num_rot_inliers = matcher.get_num_rotation_inliers()
    num_final_inliers = matcher.get_num_final_inliers()
    # NOTE(hlim): By checking the final inliers, we can determine whether
    # the registration was successful or not. The larger the threshold,
    # the more conservatively the decision is made.
    # See https://github.com/MIT-SPARK/KISS-Matcher/issues/24
    thres_num_inliers = 5
    if (num_final_inliers < thres_num_inliers):
        print("\033[1;33m=> Registration might have failed :(\033[0m\n")
    else:
        print("\033[1;32m=> Registration likely succeeded XD\033[0m\n")

    # Apply transformation to src

    print("7. Refine registrations.")
    voxel_size = 0.02
    max_correspondence_distance_coarse = voxel_size * 15
    max_correspondence_distance_fine = voxel_size * 1.5

    T = np.eye(4)
    T[:3, :3] = result.rotation
    T[0:3, 3] = result.translation

    icp_fine = o3d.pipelines.registration.registration_icp(
        source_down, target_down, max_correspondence_distance_fine,
        T, o3d.pipelines.registration.TransformationEstimationPointToPlane())

    transformed_src_o3d = src_o3d.transform(icp_fine.transformation)

    T = np.eye(4)
    T[:3, :3] = result_rl.rotation
    T[0:3, 3] = result_rl.translation

    icp_fine_rl = o3d.pipelines.registration.registration_icp(
        rlsense_down, target_down, max_correspondence_distance_fine,
        T, o3d.pipelines.registration.TransformationEstimationPointToPlane())

    transformed_rl_o3d = rlsense_o3d.transform(icp_fine_rl.transformation)
    # transformed_rl_o3d = rlsense_o3d.transform(T)

    # ------------------------------------------------------------
    # Visualization
    # ------------------------------------------------------------
    print("8. Visualize.")
    vis = o3d.visualization.Visualizer()
    vis.create_window(window_name="KISS-Matcher Viz",
                      width=1600,
                      height=1200,
                      left=300,
                      top=300)
    vis.get_render_option().point_size = 0.3
    vis.get_render_option().background_color = np.array([0, 0, 0
                                                         ])  # Black background
    vis.get_render_option().light_on = True

    # vis.add_geometry(src_o3d)
    vis.add_geometry(tgt_o3d)
    vis.add_geometry(transformed_src_o3d)
    vis.add_geometry(transformed_rl_o3d)

    vis.run()
    vis.destroy_window()

    print("Visualization complete.")
```

LimHyungTae, 2025-03-24:
Hmmm... I see. In that case, planar features from desks are too dominant compared to other measurements, which leads to incorrect registration.
If your goal is to register the RGB-D cloud on the desk, could you try filtering out the desk plane using RANSAC or something similar? That part is explained as 'Geometric suppression' in the KISS-Matcher paper.

[image]

ttsesm, 2025-03-25:
I see, well I've tested your proposal by using the [plane segmentation](https://www.open3d.org/docs/latest/tutorial/geometry/pointcloud.html#Plane-segmentation) module from open3d (quite on hand). However, I still get some quite bad performance :-(

I do not know if there is anything else that I could do to help on the feature extraction.

LimHyungTae, 2025-03-26:
[image]

As you can see above, FPFH w/ non-minimal solver works relatively worse in indoor RGB-D data, and I guess its RGB-D clouds are inconsistent. 

By the way, you don't necessarily need to perform matching solely based on point cloud, right? Then, why don't you run 2D feature extraction and matching using images and then estimate correspondences between two 3D matched points? It would be much easier.

LimHyungTae, 2025-03-26:
I tested it a few days ago, but at that moment, it does not work properly; so, I should check if the solver itself works well. 

If there is anything wrong, I would appreciate it if you could fix it and PR it.

Now this issue doesn't seem to fit the title anymore, so I'll close this issue. Feel free to open another issue if you need! :)

closed 2025-03-26 by LimHyungTae