MIT-SPARK/KISS-Matcher #40: [QUESTION] Robin as a dependency of kiss-matcher

hofbauerc, 2025-03-26:
Hi,

We are currently using KISS-Matcher in one of our projects but are a little confused on a specific CMakeLists.txt example. As depicted in the [example](https://github.com/MIT-SPARK/KISS-Matcher/blob/33aef586842c73a352b05cbce54729ec30672490/cpp/examples/CMakeLists.txt#L21C10-L21C29) on how to use in in our project, the find_package for robin has to be defined separately to kiss_matcher.

Wouldn't it be better to let robin be a dependency on kiss-matcher in order to only need [the following line](https://github.com/MIT-SPARK/KISS-Matcher/blob/33aef586842c73a352b05cbce54729ec30672490/cpp/examples/CMakeLists.txt#L22) to include kiss-matcher to a custom project. 

Or are we missing something here?

LimHyungTae, 2025-03-26:
I do agree. Then how can I fix the CMakeLists.txt to support that one? If it is doable, I'm happy to see your patch via PR.

Or you can use it like this, as presented in ros example:

```
if(EXISTS ${CMAKE_CURRENT_SOURCE_DIR}/../cpp/kiss_matcher/ AND NOT IS_SYMLINK ${CMAKE_CURRENT_SOURCE_DIR})
  add_subdirectory(${CMAKE_CURRENT_SOURCE_DIR}/../cpp/kiss_matcher ${CMAKE_CURRENT_BINARY_DIR}/kiss_matcher)
else()
  #This is what kiss_icp does.
    #cmake_minimum_required(VERSION 3.18)
    #message(STATUS "Performing out-of-tree build, fetching KISS-Matcher v${CMAKE_PROJECT_VERSION} Release from Github")
    #include(FetchContent)
    #FetchContent_Declare(
    #  ext_kiss_matcher_core PREFIX kiss_matcher_core
    #  URL https://github.com/PRBonn/kiss-matcher/archive/refs/tags/v${CMAKE_PROJECT_VERSION}.tar.gz SOURCE_SUBDIR
    #      cpp/kiss_matcher)
    #FetchContent_MakeAvailable(ext_kiss_matcher_core)
endif()
```

LimHyungTae, 2025-04-04:
As there's no further response, I'd like to close this issue.

closed 2025-04-04 by LimHyungTae