MIT-SPARK/KISS-Matcher #49: [QUESTION] How to assess that how many inliers are necessary?

beastsenpai213, 2025-04-06:
Hi LimHyungTae,

Firstly, I want to express my gratitude for your work on this point cloud match library. It has been a breeze to integrate into my localization backend. I accumulated registed scans from my frontend LIO (0.1 m voxelize)as local maps for global matching.

During the developing, I’ve noticed that the number of inliers is consistently low, ranging from 5 to 10, whereas in the examples provided, the number can reach up to 90. Despite the low inlier count, my backend still seems to be working well.

How is the inliers works and How can I assess how many inliers are necessary in real scene, if necessary ,how to improved inliers?

Thank you in advance for your assistance. :)

Here are my params:

    gicp_config:
      num_threads_: 8
      correspondence_randomness_: 20 
      max_num_iter_: 20 
      voxel_res_: 0.2 #0.2
      max_corr_dist_: 1.0 
      scale_factor_for_corr_dist_: 5.0 
      overlap_threshold_: 90.0

    kiss_config:
      voxel_size: 0.1 #0.1
      use_voxel_sampling: true
      use_quatro: false
      thr_linearity: 1.0
      num_max_corr: 5000
      ## Below params just works in general cases
      normal_r_gain: 3.0
      fpfh_r_gain: 5.0
      ## The smaller, more conservative
      robin_noise_bound_gain: 1.0
      solver_noise_bound_gain: 0.75
      enable_noise_bound_clamping: true

Here is my simple backend performance:
[image]
[image]

LimHyungTae, 2025-04-07:
Which LiDAR sensor do you use?

At a glance, the LiDAR cloud seems too spare, so you need to accumulate more scans.

Could you increase the # of accumulated frames please?

LimHyungTae, 2025-04-07:
i.e., https://github.com/MIT-SPARK/KISS-Matcher/blob/main/ros%2Fconfig%2Fslam_config.yaml#L28-L28

beastsenpai213, 2025-04-07:
(oops,I noticed that the obstacle in lab is moved)
this is how it perform when using origin pointclouds and enlarge the pointcloud window size to match(local map voxel size:0.1):

[image]

meanwhile larger pointclouds increased CPU performance consumption ,I am a little afraid that there's not enough CPU for my planning and sensing.

beastsenpai213, 2025-04-07:
I found it higher robin_noise_bound_gain and solver_noise_bound_gain could get a higher inliers.I would like to know what the impact this will have,more robust or unstable...

[image]

LimHyungTae, 2025-04-07:
Wait, you already voxelized with the size of 0.3 but set 0.1 for localization scenario? It doesn't make sense. 

If you want to localize, please accumulate the clouds for a few seconds and do localization (because there should sufficient overlap between two point clouds, but in your case, scan-to-map matching has little overlap).

If robin_noise_gan and solver_noise_bound becomes large, it might outputs more # of inliers, but it is likely to to be that they are not actually valid anymore.

LimHyungTae, 2025-04-07:
Might be also relevant to #18.

LimHyungTae, 2025-04-13:
Closed as there's no further response. Feel free to re-open this issue if you need to!

closed 2025-04-13 by LimHyungTae