MIT-SPARK/KISS-Matcher #53: [QUESTION] Can KISS-Matcher-SAM handle small rotational drift and wobble in LiDAR odometry?

BenderBot-es, 2025-04-10:
Hi, thanks a lot for the great work!

I’ve been testing KISS-Matcher-SAM along with a LiDAR odometry system based on FAST-LIO (with a rotating 16-channel LiDAR), but I haven’t been able to get good results. My odometry has some rotational vibrations, and the system doesn’t seem to correct them at all. The results I get look pretty much like just accumulating the raw pointclouds from FAST-LIO.

Is submap-to-submap registration being performed even when there are no loop closures? From what I’ve read in the documentation, it seems like it should — but I’m not seeing any improvement in my tests.

Also, loop closures seem quite heavy in terms of compute. Is there any specific hardware requirement or recommendation?

Thanks again for the great work, and I’d really appreciate any advice or insight on these questions!

LimHyungTae, 2025-04-13:
I kind of understood, but could you describe it in more detail with a screenshot? 

I'm particularly curious why rotational vibration is assumed to always exist. I think that could be handled by LIO.

In addition, if you set larger voxelization, local drift can also be somewhat covered. I'm sure you missed tuning some parameters. (because the MIT campus example is also acquired by Velodyne 16 channel LiDAR with a poor quality IMU sensor.)

LimHyungTae, 2025-04-13:
Sorry, I wrongly pressed the button.

LimHyungTae, 2025-04-23:
As there's no response anymore, I'd like to close it, but feel free to re-open :)

closed 2025-04-23 by LimHyungTae