MIT-SPARK/KISS-Matcher #54: [QUESTION]

dhfsu, 2025-04-15:
\*\* PLEASE READ AND DELETE MY MESSAGE\*\*

Hi, this is Hyungtae, the first author of KISS-Matcher. I appreciate you visiting our Github issue. I always want to help you as much as I can; however, due to my limited time for research, I may not be able to thoroughly address every issue. I kindly ask for your understanding.

For everyone’s benefit, please make your questions **as detailed as possible**. Please avoid very short, one-line questions, because they often require multiple follow-up inquiries for clarification, which can be time-consuming and stressful.

To ensure smoother communication, kindly include **as many relevant details as** you can when submitting your issue. Thank you for your cooperation.

## In sum, Please avoid the following when submitting an issue:

### Bad case A. One-line or vague questions

Example: "It doesn't work. What should I do?"
Instead, provide details such as error messages, steps to reproduce, system specifications, and expected vs. actual behavior.

### Bad case B. Requesting general debugging without context

Example: "Can you check my code?" without explaining the problem, expected outcome, or attempted solutions.

### Bad case C. Posting duplicate issues

Before submitting, check if a similar issue has already been raised and addressed.

### Bad case D. Unrelated questions

## Please keep issues relevant to KISS-Matcher. For general research questions, consider other forums such as Stack Overflow or related academic discussions.

**Have you read the documentation?**

- [ ] Yes
- [x] No - then this issue will be closed.

LimHyungTae, 2025-04-16:
That's interesting! B

y the way, is the radar data in 3D? Since FPFH is sensitive to point density, I wonder if it might not work well when the radar data is too sparse.

dhfsu, 2025-04-16:
Yes, the point cloud data is 3D point cloud collected by the Livox Mid360. There are approximately more than 20,000 points in one frame. When we take one frame of data and match it with the entire map, the problem of matching failure will occur. I would like to ask whether it is due to the relatively small number of points in one frame of data。

[image]

LimHyungTae, 2025-04-16:
Please read #18 and #50

LimHyungTae, 2025-04-16:
> ## 💡 Why Sufficient Overlap Matters in Point Cloud Registration
> In point cloud registration, the goal is to align two sets of 3D points (called source and target) into a common coordinate frame. This process relies on finding corresponding points between the two point clouds that represent the same physical structure.
> 
> Let’s take a look at the image above:
> 
> [image]
> 
> * 🩵 The cyan region is the target point cloud.
> * ⚪ The gray region is the source point cloud.
> * 🟧 The orange region is the source after registration (i.e., the estimated transformation applied).
> 
> As you can see, the gray source cloud and the cyan target cloud barely overlap. Because of this limited overlap, the algorithm struggles to find enough valid correspondences — pairs of points that should match between the two clouds. Especially, in case where the 🩵 target cloud is much wider, it inevitably outputs spurious correspondences. As a result, the 🟧 estimated cloud ends up in the wrong place.
> 
> This is a very natural failure mode in point cloud registration. No matter how good the algorithm is, if there’s not enough shared geometry (i.e., overlapping area) between the input clouds, registration is very likely to fail.
> 
> To summarize:
> 
> > Sufficient overlap is essential for successful registration. When two point clouds don’t share enough common structure, the algorithm cannot reliably estimate the correct transformation.

LimHyungTae, 2025-04-23:
As there's no response anymore, I'd like to close it, but feel free to re-open :)

closed 2025-04-23 by LimHyungTae