MIT-SPARK/KISS-Matcher #55: [QUESTION] Specify custom install directory

JeffR1992, 2025-04-18:
I'm loving KISS-Matcher's performance (both in terms of speed and accuracy), but have a question regarding the installation setup.

Using the default `Makefile`, is it possible to install KISS-Matcher and its dependencies into a custom directory instead of into system paths? I couldn't seem to find such an option by default, so had to modify the `cppinstall` and `cppinstall_matcher_only` sections as follows:

```
cppinstall: deps ascii_art
	@mkdir -p cpp/kiss_matcher/build
	@cmake -Bcpp/kiss_matcher/build cpp/kiss_matcher -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=$(INSTALL_DIR)
	@cmake --build cpp/kiss_matcher/build -j$(nproc --all)
	@$(SUDO) cmake --install cpp/kiss_matcher/build
	@$(SUDO) cmake --install cpp/kiss_matcher/build/_deps/robin-build

cppinstall_matcher_only: ascii_art
	@mkdir -p cpp/kiss_matcher/build
	@cmake -Bcpp/kiss_matcher/build cpp/kiss_matcher -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=$(INSTALL_DIR)
	@cmake --build cpp/kiss_matcher/build -j$(nproc --all)
	@$(SUDO) cmake --install cpp/kiss_matcher/build
```

So I could then build and install using:

`make cppinstall INSTALL_DIR=/path/to/install/dir`

However, if an option to provide an install prefix path is already available, guidance on how to use it would be appreciated. Thanks!

LimHyungTae, 2025-04-18:
Hmm...that's a great point. Then, how can we link the local lib directory?

LimHyungTae, 2026-09-29:
I think this issue might be resolved now (hopefully).

closed 2026-09-29 by LimHyungTae