MIT-SPARK/KISS-Matcher #57: [QUESTION] LC rejected. KISS-Matcher failed !

Thetrafficlights, 2025-04-26:
Dear author, what kind of problem occurred when I was using your kiss_matcher? Could you help me see what might have caused it?

[kiss_matcher_sam-1] [INFO] [1745645989.378110549] [km_sam.kiss_matcher_sam]: Execute coarse-to-fine alignment: # src = 1572, # tgt = 1802
[kiss_matcher_sam-1] [WARN] [1745645989.568955410] [km_sam.kiss_matcher_sam]: # final inliers: 0 < 50
[kiss_matcher_sam-1] [WARN] [1745645989.569013466] [km_sam.kiss_matcher_sam]: LC rejected. KISS-Matcher failed
[kiss_matcher_sam-1] [INFO] [1745645989.569026876] [km_sam.kiss_matcher_sam]: Reg: 193.0 msec

LimHyungTae, 2025-04-26:
As you saw, it has certifiable characteristics (in case of LC, rejecting false loops is very very important for precise mapping). 

Which point clouds do you use now? In general, given normal cases, `# of final inliers` should be over 50.

Thetrafficlights, 2025-04-27:
Does "which point clouds" refer to the point cloud input of kiss-matcher? My original point cloud data (XYZIRT) was a package recorded by my own sensor. Since spark-fast-lio does not support this lidar sensor, I converted the point cloud format to velodyne (still XYZIRT, but the T values were different. The current time of the laser point -> the time offset relative to the first point.) Then, through spark-fast-lio, the output topic is /cloud_registered laser point cloud, that is, the input of kiss-matcher -scan_topic

Thetrafficlights, 2025-04-27:
我spark-fast-lio存在这样的警告，我不知道是否和它有关
[spark_lio_mapping-1] [WARN] [1745716677.946621101] [lio_mapping]: meas.lidar->points.back().curvature (49.98) should be close to 100. Please check the `timestamp_unit` or values of `time` (or `t`) field of the point cloud input from your sensor.
顺便附上我kiss-matcher的config.yaml
  ros__parameters:
    loop_pub_hz: 0.1                 
    loop_detector_hz: 2.0            
    loop_nnsearch_hz: 2.0            
    loop_pub_delayed_time: 30.0      
    map_update_hz: 0.2               
    vis_hz: 1.0                      

    store_voxelized_scan: false      
    voxel_resolution: 0.3            
    map_voxel_resolution: 1.0        
    save_voxel_resolution: 0.3       

    keyframe:
      keyframe_threshold: 1.5        
      num_submap_keyframes: 1       

    loop:
      verbose: true                 
      is_multilayer_env: false               
      loop_detection_radius: 20.0            
      loop_detection_timediff_threshold: 90.0

    local_reg:
      num_threads: 12                 
      correspondences_number: 20      
      max_num_iter: 32                
      scale_factor_for_corr_dist: 5.0 
      overlap_threshold: 55.0         
                                                                 
    global_reg:
      enable: true
      num_inliers_threshold: 50       

    result:
      save_map_pcd: true
      save_map_bag: false
      save_in_kitti_format: false
      seq_name: "sequence"

Thetrafficlights, 2025-04-27:
I have such a warning in spark-fast-lio. I don't know if it is related to it:
[spark_lio_mapping-1] [WARN] [1745716677.946621101] [lio_mapping]: meas.lidar->points.back().curvature (49.98) should be close to 100. Please check the timestamp_unit or values of time (or t) field of the point cloud input from your sensor.

By the way, here is the config.yaml of my kiss-matcher:
loop_detector_hz: 2.0
loop_nnsearch_hz: 2.0
loop_pub_delayed_time: 30.0
map_update_hz: 0.2
vis_hz: 1.0

store_voxelized_scan: false      
voxel_resolution: 0.3            
map_voxel_resolution: 1.0        
save_voxel_resolution: 0.3       

keyframe:
  keyframe_threshold: 1.5        
  num_submap_keyframes: 1       

loop:
  verbose: true                 
  is_multilayer_env: false               
  loop_detection_radius: 20.0            
  loop_detection_timediff_threshold: 90.0

local_reg:
  num_threads: 12                 
  correspondences_number: 20      
  max_num_iter: 32                
  scale_factor_for_corr_dist: 5.0 
  overlap_threshold: 55.0         
                                                             
global_reg:
  enable: true
  num_inliers_threshold: 50       

result:
  save_map_pcd: true
  save_map_bag: false
  save_in_kitti_format: false
  seq_name: "sequence"

Thetrafficlights, 2025-04-27:
Another point is that in spark-fast-lio, I found that the parameter of my feature_enabled was in the false state, but the type of my lidar was mechanical. I guess this is what you want to ask

LimHyungTae, 2025-04-27:
1. You should set the processing part for your own sensor. As an example, see this part (if you use a special sensor, then you might parse the data for your sensor):

https://github.com/MIT-SPARK/spark-fast-lio/blob/e48354c90664410b30568911926f1449001c5ec6/spark_fast_lio/config/velodyne.yaml#L11

2. What I'm asking is how the point cloud is visualized in rviz. Please help me understand as much information as possible when you ask me questions. As a quick solution, you can increase `num_submap_keyframes` and see how it goes.

Thetrafficlights, 2025-04-27:
Thank you for your reply. From the perspective of the point cloud visualization interface, this problem occurs every time when my car is about to reach the starting point and the distance is within the radius of the loop detection. I'll try to increase the value of num_submap_keyframes first

Thetrafficlights, 2025-04-27:
It's still the same error report

LimHyungTae, 2025-04-27:
I strongly believe it's not the problem of KISS-Matcher/SPARK-FAST-LIO. 

And I can't help you because you just said 'it does not work' without any screenshots and results. 

What am I supposed to do? Please show

1. Show spark-fast-lio rviz
- Did you set the extrinsic of R and t properly?

2. Show KISS-Matcher pipeline rviz. Currently, the number of point clouds is too few.

Just run examples that I provided and analyze by yourself first. The Git issue is not for just a Q&A session bur for discussion to enhance usability/problems.

I also have narrow bandwidth :(

LimHyungTae, 2025-04-28:
Please read https://github.com/MIT-SPARK/KISS-Matcher/issues/18 and https://github.com/MIT-SPARK/KISS-Matcher/issues/50

Thetrafficlights, 2025-04-28:
Sorry, I originally thought that github could only reply carefully with text messages. Now I'll send you my screenshot:
1.kiss_matcher rviz
[image]
2.kiss_matcher terminal
[image]
3.spark_fast_lio rviz
[image]

LimHyungTae, 2025-04-28:
Man, it's not the problem of KISS-Matcher, isn't it (if I understand correctly, it just broke)?

First, read this guide (https://github.com/MIT-SPARK/KISS-Matcher/blob/main/ros/README_ROS2_REGISTRATION.md), run the registration in ROS2, and check whether it works properly. If registration works correctly, then it's not  the problem of KISS-Matcher.

Also, please report your thoughts on the error. The issue system is not just for Q&A — it's meant for discussion.

Thetrafficlights, 2025-04-28:
The subsequent program reported errors because there was no normal loop, resulting in the failure to publish the visual loop results

[image]

LimHyungTae, 2025-05-28:
Any updates? Feel free to reopen the issue if you need.

closed 2025-05-28 by LimHyungTae