MIT-SPARK/KISS-Matcher #63: [QUESTION] relocalization and 6-DOF pose extraction

shubham-shahh, 2025-05-27:
Hi,
Thank you so much for this awesome work. I have 2 questions:
1) I have a 3D map of an environment, and I am willing to navigate in the same environment and relocalize myself using the live scan data and the existing 3D map. Is there a way to extract a 6 DOF pose from the scan matcher?

2) If I am using a different sensor to map and a different 3D sensor to relocalize, what sort of inaccuracies/artifacts can I expect? I'd love to know your thoughts.

Thanks!
**Have you read the documentation?**

- [X] Yes
- [] No - then this issue will be closed.

LimHyungTae, 2025-05-28:
Hi @shubham-shahh, thank you for your interest.

1. Yes, I'd like to share a pseudo-code as follows:

```
a. Load a 3D point cloud with voxel size v  
b. Run the LiDAR odometry algorithm  
c. Wait until a submap is sufficiently accumulated by the LiDAR odometry (i.e., two clouds should have sufficient overlaps) 
d. Run KISS-Matcher and check the final number of inliers  
e. If the number of final inliers is sufficiently large, perform relocalization
f. Repeat steps c–e at regular time intervals to correct poses continuously  
```

2. With scan-to-map, it cannot be registered, but as long as you follow step c, then the cloud points are uniformly distributed, which enables running KISS-Matcher.

LimHyungTae, 2025-05-28:
Read #18 and #50

LimHyungTae, 2025-05-28:
For re-localization, you can drectly exploit this code: https://github.com/MIT-SPARK/KISS-Matcher/blob/main/ros/src/inter_frame_alignment.cpp