MIT-SPARK/KISS-Matcher #76: [BUG] Cannot build for python

timfanselow-bc, 2025-08-07:
I'm trying to build the python package as per README instructions. 

`pip install -e python/` is failing with error:

```
      CMake Error at CMakeLists.txt:4 (cmake_minimum_required):
        Compatibility with CMake < 3.5 has been removed from CMake.
      
        Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
        to tell CMake that the project requires at least <min> but has been updated
        to work with policies introduced by <max> or earlier.
      
        Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.
      
      
      CMake Error at /tmp/tmpp7x6u1xt/build/_deps/robin-src/cmake/DownloadProject.cmake:171 (message):
        CMake step for pmc failed: 1
      Call Stack (most recent call first):
        /tmp/tmpp7x6u1xt/build/_deps/robin-src/cmake/DownloadExternal.cmake:17 (download_project)
        /tmp/tmpp7x6u1xt/build/_deps/robin-src/CMakeLists.txt:51 (robin_download_pmc)
```

I see this is noted in https://github.com/MIT-SPARK/KISS-Matcher/issues/64, but that ticket was resolved by updating robin - I'm thinking maybe this is a new issue.

This is my dockerfile reproducer
```
FROM ubuntu:noble
ARG DEBIAN_FRONTEND=noninteractive
RUN apt-get update
RUN apt-get install -y python3-pip python3-venv git
RUN git clone https://github.com/MIT-SPARK/KISS-Matcher.git
WORKDIR KISS-Matcher
RUN python3 -m venv venv
RUN . venv/bin/activate \
&& pip install --upgrade pip setuptools wheel scikit-build-core ninja cmake build
RUN . venv/bin/activate \
&& pip install -e python/
```

LimHyungTae, 2025-08-08:
Yep that issue might be too high CMake version. You might have use 4.xx version - could you please check it?

manavkulshrestha, 2025-09-14:
I'm running into this same error even with cmake 3.31.6. Any fix yet?
```
-- Install KISS-Matcher using local C++ files...

  ██╗  ██╗██╗███████╗███████╗
  ██║ ██╔╝██║██╔════╝██╔════╝
  █████╔╝ ██║███████╗███████╗█████╗
  ██╔═██╗ ██║╚════██║╚════██║╚════╝
  ██║  ██╗██║███████║███████║
  ╚═╝  ╚═╝╚═╝╚══════╝╚══════╝

  ███╗   ███╗ █████╗ ████████╗ ██████╗██╗  ██╗███████╗██████╗
  ████╗ ████║██╔══██╗╚══██╔══╝██╔════╝██║  ██║██╔════╝██╔══██╗
  ██╔████╔██║███████║   ██║   ██║     ███████║█████╗  ██████╔╝
  ██║╚██╔╝██║██╔══██║   ██║   ██║     ██╔══██║██╔══╝  ██╔══██╗
  ██║ ╚═╝ ██║██║  ██║   ██║   ╚██████╗██║  ██║███████╗██║  ██║
  ╚═╝     ╚═╝╚═╝  ╚═╝   ╚═╝    ╚═════╝╚═╝  ╚═╝╚══════╝╚═╝  ╚═╝

  CMake Deprecation Warning at /home/csuser/KISS-Matcher/cpp/kiss_matcher/3rdparty/find_dependencies.cmake:25 (cmake_policy):
    The OLD behavior for policy CMP0135 will be removed from a future version
    of CMake.

    The cmake-policies(7) manual explains that the OLD behaviors of all
    policies are deprecated and that a policy should be set to OLD only under
    specific short-term circumstances.  Projects should be ported to the NEW
    behavior and not rely on setting a policy to OLD.
  Call Stack (most recent call first):
    /home/csuser/KISS-Matcher/cpp/kiss_matcher/CMakeLists.txt:43 (include)

  CMake Deprecation Warning at /tmp/tmp0m48gxi5/build/_deps/tbb-src/CMakeLists.txt:15 (cmake_minimum_required):
    Compatibility with CMake < 3.10 will be removed from a future version of
    CMake.

    Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
    to tell CMake that the project requires at least <min> but has been updated
    to work with policies introduced by <max> or earlier.

  -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
  -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
  -- Found Threads: TRUE
  -- Checking for one of the modules 'hwloc'
  -- Looking for getcontext
  -- Looking for getcontext - found
  -- IPO enabled
  -- GNU Assembler version: 2.42  (2042)
  -- The tbbbind_2_5 target will be configured using the HWLOC 2.10.0
  CMake Warning (dev) at /tmp/pip-build-env-kk_pdqmf/normal/lib/python3.10/site-packages/cmake/data/share/cmake-4.1/Modules/FetchContent.cmake:1953 (message):
    Calling FetchContent_Populate(robin) is deprecated, call
    FetchContent_MakeAvailable(robin) instead.  Policy CMP0169 can be set to
    OLD to allow FetchContent_Populate(robin) to be called directly for now,
    but the ability to call it with declared details will be removed completely
    in a future version.
  Call Stack (most recent call first):
    /home/csuser/KISS-Matcher/cpp/kiss_matcher/3rdparty/robin/robin.cmake:31 (FetchContent_Populate)
    /home/csuser/KISS-Matcher/cpp/kiss_matcher/3rdparty/find_dependencies.cmake:36 (include)
    /home/csuser/KISS-Matcher/cpp/kiss_matcher/3rdparty/find_dependencies.cmake:42 (find_external_dependency)
    /home/csuser/KISS-Matcher/cpp/kiss_matcher/CMakeLists.txt:43 (include)
  This warning is for project developers.  Use -Wno-dev to suppress it.

  CMake Deprecation Warning at /tmp/tmp0m48gxi5/build/_deps/robin-src/CMakeLists.txt:3 (cmake_minimum_required):
    Compatibility with CMake < 3.10 will be removed from a future version of
    CMake.

    Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
    to tell CMake that the project requires at least <min> but has been updated
    to work with policies introduced by <max> or earlier.

  CMake Error at CMakeLists.txt:4 (cmake_minimum_required):
    Compatibility with CMake < 3.5 has been removed from CMake.

    Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
    to tell CMake that the project requires at least <min> but has been updated
    to work with policies introduced by <max> or earlier.

    Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.

  CMake Error at /tmp/tmp0m48gxi5/build/_deps/robin-src/cmake/DownloadProject.cmake:171 (message):
    CMake step for pmc failed: 1
  Call Stack (most recent call first):
    /tmp/tmp0m48gxi5/build/_deps/robin-src/cmake/DownloadExternal.cmake:17 (download_project)
    /tmp/tmp0m48gxi5/build/_deps/robin-src/CMakeLists.txt:51 (robin_download_pmc)

  -- Configuring incomplete, errors occurred!

  *** CMake configuration failed
  error: subprocess-exited-with-error
  
  × Building editable for kiss_matcher (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  full command: /home/csuser/anaconda3/envs/km310/bin/python3.10 /home/csuser/anaconda3/envs/km310/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py build_editable /tmp/tmpoxgf82ys
  cwd: /home/csuser/KISS-Matcher/python
  Building editable for kiss_matcher (pyproject.toml) ... error
  ERROR: Failed building editable for kiss_matcher
Failed to build kiss_matcher
error: failed-wheel-build-for-install

× Failed to build installable wheels for some pyproject.toml based projects
╰─> kiss_matcher
(km310) csuser@ideas12:~/KISS-Matcher$ which cmake && cmake --version   # should print 3.31.6
/home/csuser/anaconda3/envs/km310/bin/cmake
cmake version 3.31.6
```